bolderflight/sbus #5: Channel 17 and 18

onelittlegoat, 2017-11-05:
Hey Brian,

Firstly, great work :)  Secondly, what is the purpose of channels 17 and 18 and how and what do you write to them? 

Cheers,

Phil

flybrianfly, 2017-11-06:
Good question. I included it because I saw it documented in [Uwe Gartmann's code](https://os.mbed.com/users/Digixx/notebook/futaba-s-bus-controlled-by-mbed/). I'm busy with some other software at the moment, which has a short deadline, but let me take a look and get back to you in a week or so.

onelittlegoat, 2017-11-06:
Hey Brian,

 

Thanks for getting back to me.  I’m using a hacked version of various other libraries but figured out how to set the flags.  I found your library while trying to get a better understanding.  

 

The following does the trick;

 

Package[23] |= (1<<1) 

 

😊.  It took me a while to realise the bit order is reversed.  I was stuck for ages on why |= (1 << 6) wasn’t working.

 

Cheers,

 

Phil

flybrianfly, 2021-02-19:
Added CH17 and CH18 support.

closed 2021-02-19 by flybrianfly