bolderflight/sbus #9: don't work on STM32F103C8T6

dirUAV, 2018-09-19:
Hi
I need to handler sbus signal to continue my project by stm32f103c8t6. I use your examples, but it wasn't successful! Arduino compiler shows C:\Users\user\AppData\Local\Temp\arduino_build_27820/SBUS_example.ino.elf section `.text' will not fit in region `rom'

flybrianfly, 2018-09-19:
This library works with Teensy 3.x and LC devices, the STM32L4, and the Maple Mini. I would recommend working with the writer of the Arduino core that STM32F103C8T6 uses to see what's needed to integrate this library. The biggest task is usually getting the core library to support an inverted logic 100000 baud hardware serial. I'd be happy to merge pull requests supporting it once the core modifications are made.

closed 2018-09-19 by flybrianfly

dirUAV, 2018-09-20:
It's true! actually i read sbus signal by arduino due and a simple converter (Transistor 2N3904 and two resistors). Firstly you said "Receiver signal should be connected to the microcontroller RX pin on the specified hardware serial port.". Is it possible? Secondly, how can i receive the sbus signal by STM32F103C8T6 and interrupt?