bolderflight/sbus #13: Question.  channels[0] to channels[1] ?

user1321, 2019-01-31:
Good Day!
Please, correct me if I am wrong, but should that part of code:

packet[1] = (uint8_t) ((channels[0] & 0x07FF));
        packet[2] = (uint8_t) ((channels[0] & 0x07FF)>>8 | (channels[1] & 0x07FF)<<3);
        packet[3] = (uint8_t) ((channels[1] & 0x07FF)>>5 | (channels[2] & 0x07FF)<<6);
packet[4] = (uint8_t) ((channels[2] & 0x07FF)>>2);

should be like that:

packet[1] = (uint8_t) ((channels[0] & 0x07FF));
        packet[2] = (uint8_t) ((channels[1] & 0x07FF)>>8 | (channels[2] & 0x07FF)<<3);
        packet[3] = (uint8_t) ((channels[2] & 0x07FF)>>5 | (channels[3] & 0x07FF)<<6);
packet[4] = (uint8_t) ((channels[3] & 0x07FF)>>2);

Thank you.

flybrianfly, 2019-01-31:
No, the channels are 11 bits wide and you're stuffing them into an array of 8 bit containers. You can see a depiction of how the channels are packed in the array here:
https://github.com/bolderflight/SBUS/blob/master/extras/bit-mapping.pdf

user1321, 2019-01-31:
Oh, I see.
Sorry.

closed 2019-01-31 by user1321