bolderflight/sbus #30: Failing to recieive

arbrog, 2020-03-05:
Using an Arduino Mega 2560 with a FrySky R-XSR. I am using the SBUS_example.ino (as a base) on Serial2 (Pin 17 RX) and verified the program starts and the main loop runs but I never get inside the if
```
if(x8r.read(&channels[0], &failSafe, &lostFrame)){

    Serial.print("working");
  }
```
I put some return statements to shortcircuit SBUS read and it looks like the `if (parse()) {` on line 66 of SBUS.cpp is returning false.
I used an oscilloscope to verify that my inverting circuit is working so I'm not sure why the packet isn't coming through correctly. 
```
#include "SBUS.h"

// a SBUS object, which is on hardware
// serial port 1
SBUS x8r(Serial2);

// channel, fail safe, and lost frames data
uint16_t channels[16];
bool failSafe;
bool lostFrame;

void setup() {
  // begin the SBUS communication
  x8r.begin();
  Serial.begin(115200);
  Serial.println("start");
  

}

void loop() {

  // look for a good SBUS packet from the receiver
  Serial.println("polling: ");
  if(x8r.read(&channels[0], &failSafe, &lostFrame)){

    Serial.print("working");
  }
  
}
```

arbrog, 2020-03-05:
so further debugging, in parse() I never get into `while (_bus->available() > 0) {` on line 299 so there are never bytes available to read?

swampd32618, 2020-04-15:
Did you ever resolve this issue? I'm experiencing the same problem and it's incredibly frustrating!

arbrog, 2020-04-15:
I did not, but I also haven't worked on the project this was for since due to COVID

khanhln2907, 2020-07-27:
I tried printing all values received from UART Buffer and they were wrong (in comparision to another app I used such as Logic Analyzer or Hterm). I guessed there might be something wrong with the the baudrate or format during the serial initialization so I modified the "begin()" of SBUS by just using "_bus->begin(_sbusBaud);" and it worked.

circuitBurn, 2020-12-27:
Any update from anyone? Trying to get an R8 PRO receiver working, running into the same issue and @khanhln2907's suggestion didn't change anything for me.

flybrianfly, 2021-02-15:
If anyone is able to get to the bottom of this, please re-open the issue with the specific steps taken, the microcontroller tested, and the necessary code changes relative to the most recent code. Thanks!

closed 2021-02-15 by flybrianfly