bolderflight/sbus #31: How to add Compatibility with the Arudino Micro and Arudino Leonardo

mschindler301, 2020-03-20:
I have found a way to add compatibility with the Arduino Micro and Arduino Lenardo (ATMega32u4 based boards). According to the [Arduino Serial Reference](https://www.arduino.cc/en/pmwiki.php?n=Reference/serial), the ATMega32u4 uses Serial1 as its main and only Serial port.

In SBUS.cpp, I added this line to the end of the SBUS::begin()

`_bus->begin(_sbusBaud, SERIAL_8E2); //For ATMega32u4`

And it works! That is the only modification you need to make to the library to make this compatible with ATMega32u4 boards, and I thought this would be a great addition to the library to support more boards.

flybrianfly, 2021-02-15:
Added.

closed 2021-02-15 by flybrianfly

robomaniac, 2022-03-02:
ah, in #54 you mention older version

> There's some older versions older versions of this library that support AVR; check the checklog. 

do you need command line to check the logs? that is new to me.
I did trial and error and it's 1.0.1 that compile with arduino. I still wish to learn how to check logs.