bolderflight/sbus #39: Trying to use library to connect to Pixhawk Cube Black SBus Out

Officepot, 2020-09-22:
As the title says, currently works great with my futaba reciever but want to take channels out of pixhawk and into a arduino. Currently comes in like this 
[image]. All stick movement only changes channel one, all other channels stay the same.

flybrianfly, 2020-09-22:
Any way to test what the PixHawk is sending? Seems like it could be a PixHawk configuration issue.

Officepot, 2020-09-22:
I have went through and tried to use the sendbyte() using arduino. Not sure how many bytes total the pixhawk is sending. The above is what channel 1-5 output to serial using the SBUS receive. I have posted on the forum for the pixhawk 2 cube black and asked for the SBUS pixhawk configuration. I havent seen anyone use the SBUS out for anything other than frsky or futaba sbus servos so no one has ever asked what is actually being put out. I have a SBUS decoder and plugging that into the Pixhawk works fine.

Officepot, 2020-09-24:
Figured it out, seems we were needing ReadCal() for the futaba reciever where the Pixhawk SBUS out just needed read(). Thanks for the help!

closed 2020-09-24 by Officepot