bolderflight/sbus #41: Teensy 3.2 - Reading from receiver & sending to USB -> 'deadlock' ?!

mw46d, 2020-10-12:
A very simple test program seems to 'deadlock' [for lack of better understanding yet] mt Teensy 3.2. The Arduino Serial Monitor does reports a couple of lines (around eight) and then does not report new readings and it even has problems, closing. The Teensy does not like to accept a new program either afterwards:-( I usually disconnect the receiver and download a Blink sketch in between:-(
[SBUS_t1.zip](https://github.com/bolderflight/SBUS/files/5363101/SBUS_t1.zip)

mw46d, 2020-10-12:
It looks like things get out of sync?! It works fine when I do things in the following order
* Power on the Teensy 3.2 & receiver
* Start the serial monitor (not showing anything at that point)
* Power on the transmitter and select the correct model
--> The serial monitor spits out lines as expected. I watched it for 30 seconds or so (much longer than above)
It does not work, when the transmitter is on, when the Teensy/receiver are switched on. Then I only see a couple of lines:-(

flybrianfly, 2021-02-15:
Should be resolved, re-open the issue if you're still having problems with v2.0.0 of this library.

closed 2021-02-15 by flybrianfly