bolderflight/sbus #46: Arduino ... another problem (error)

farmaceuta777, 2021-02-19:
Hello again ... during compilation I get an error..SbusRx / SbusTx does not name a type ... why? I would like to ask for an explanation as a child because I am very beginner ... best regards :-)
(I added the etl library!)
[image]

flybrianfly, 2021-02-19:
My best guess is that the sbus-arduino library wasn't installed correctly. Is it in the Arduino/libraries directory? If you downloaded the zip, did you rename the folder to sbus-arduino?

farmaceuta777, 2021-02-19:
the library is in the Arduino / library folder.. it is called "sbus-arduino-main"
now tried to rename it to "sbus-arduino" .. no change.
I uninstalled and reinstalled and still the same error .. "not name a type" ...

flybrianfly, 2021-02-19:
After you rename it to "sbus-arduino" you might need to close and restart Arduino so that it refreshes its list of libraries.

farmaceuta777, 2021-02-19:
Unfortunately no changes ... after renaming and closing the IDE multiple times, the same problem occurs ... :-(

flybrianfly, 2021-02-19:
At this point, I'm not sure what to try. It compiles fine for me when targeting that board. You may want to ask on some Arduino forums.

farmaceuta777, 2021-02-19:
Ok ok :-)
I will try to update my Arduino IDE version to the newest one and see if it will change something ... I will also ask on the forums ...
Thank you very much for your help and the time you gave me! :-)
greetings :-)

farmaceuta777, 2021-02-19:
Ok ... my IDE version is 1.8.8
now I downloaded the latest 1.8.13
and surprisingly it compiles normally! :-)
after re-compiling on IDE 1.8.8 it also works! ...
is it possible to run a link on a software Uart? (SoftwireSerial library)?
I need to change the hardwareserial library to softwareserial in function and it will work?

farmaceuta777, 2021-02-19:
Unfortunately it does not work ... is it possible to use pins other than the built-in Uart?

flybrianfly, 2021-02-19:
No, it only works with the HardwareSerial.

closed 2021-02-19 by flybrianfly