bolderflight/sbus #47: update package

dgorissen, 2021-04-15:
First of all thanks for developing a handy lib. I've been using this on a teensy 4 + x8r successfully by installing the package through platformio.

I ran into a bug today and after checking with the github repo to find a fix, I noticed that I was running version 1.0.1 which is pretty old. However its the latest version that is available through the platformio package manager:

https://platformio.org/lib/show/5622/Bolder%20Flight%20Systems%20SBUS

Is there any chance of updating the package to the current release?

flybrianfly, 2021-04-15:
I would chat with the Platform IO folks and point them to [this repo](https://github.com/bolderflight/sbus-arduino/). I'm not sure how packaging is done in their system.

closed 2021-04-15 by flybrianfly

dgorissen, 2021-04-15:
Thanks for the speedy reply. Turned out it was easy. I set this in my platformio.ini file and it all worked:

```
lib_deps = 
	https://github.com/bolderflight/sbus-arduino/archive/refs/tags/v2.1.2.zip

```

Riffer, 2021-04-24:
After fiddling around with the PlatformIO console and getting information from their website, it seems they expect the maintainer of the library to register new versions. I was able to do it for myself with a current copy of the source code but now this library 'belongs' to my personal path, even all other information is kept intact: https://platformio.org/lib/show/12069/Bolder%20Flight%20Systems%20SBUS

Somewhen  @flybrianfly  must have done the same steps because formerly the library path was at bolder flight.