bolderflight/sbus #48: SBUS2

nightfury-11, 2021-07-01:
this library is compatible with SBUS2? or just SBUS?

Second, if SBUS2 is not supported, how are you sending data to the transmitter?

flybrianfly, 2021-07-01:
This library supports receiving actuator commands from a SBUS or SBUS2 receiver and sending commands to SBUS or SBUS2 servos. It does not support the telemetry aspects of SBUS2.

closed 2021-07-01 by flybrianfly

nightfury-11, 2021-07-02:
does that mean I can record frameloss onto lets say an SD card?

flybrianfly, 2021-07-02:
Yes, see the lost_frame and failesafe methods in the API.

nightfury-11, 2021-07-03:
Understood. 

I had a couple questions about the schematic. 

The RX pin will act as the "signal" wire correct? since we are only receiving data and not transmitting it. 

The wire from the arduino will go into the DATA port on the receiver, or the SBUS port? 

if I'm using an SBUS2 receiver, example R7008SB, should I connect it to the SBUS or SBUS2 port? 

Lastly, since I'll need a serial inverter, can you advise on how I should go about making one?

flybrianfly, 2021-07-03:
Yes, SBUS signal to the RX pin. I haven't used a Futaba Receiver (just received feedback on making code compatible with them), so you'd have to try both ports and see which would work. I would just search online for inverting circuits, I've only used microcontrollers that support the inverted serial natively.

nightfury-11, 2021-07-06:
Ok, I'll try both ports. How will I know which one is working and which one isn't? Via the serial monitor?