bolderflight/sbus #51: FrSky Receiver Range

Peter010103, 2021-12-10:
Hi, thanks for creating an easy to use library for outputting SBUS signals.

I just had a question on the range values of 172-1811 offset values from the Readme. 1 channel consists of 11 bits (0-2047) and the library uses the extended range of -150% ~ +150%, which would correspond to PWM values of 750-2250 microseconds. 

Doing a linear mapping to 1000-2000 microseconds, wouldn't the required range for outputting these standard rx PWM values be 341-1705. How did you choose the range 172-1811? Could this be specified in the Readme?

Thanks !

flybrianfly, 2021-12-10:
SBUS is a different protocol than PWM. Whereas PWM uses an analog pulse with a duration of 1000 to 2000 us, SBUS uses a digital byte stream with 11 bits per channel. The 172 - 1811 is simply the range of digital values per channel sent by FrSky using the default -100 to +100 percent range. Changing the range in the FrSky transmitter to +/-150% enables the full 0 - 2047 range. This library doesn't do any SBUS to PWM or PWM to SBUS conversion.

closed 2021-12-10 by flybrianfly

Peter010103, 2021-12-11:
Ok thanks for your response. If there are 11 bits that can take values of 0-2047, why not write digital outputs of the "what would have been" PWM values to the digital representation as the 1000-2000 range lies within this interval? 

As in, wouldn't having a range of 1639 (=1811-172) representing values in the range 1000-2000 be reducing resolution?

Sorry if this is unrelated to this library which generates the required signals. But I'm just a bit confused as to why SBUS does this and feel there is something I'm missing :)

flybrianfly, 2021-12-12:
1. I don't think comparing SBUS to PWM is the correct approach, they are completely different protocols and treated as such.
2. I'm not sure why FrSky chose 172 - 1811 as their default range with a need to change settings to get the full 0 - 2047 range. I usually change the setting to get the full 0 - 2047 range anyway, in order to have higher resolution from pilot commands. I only include it in the README so people trying out the library with an FrSky receiver aren't surprised when it is not giving the full 11-bit range.