bolderflight/sbus #54: Compile Error with Arduino

martinglass, 2022-02-27:
Trying to use this repo with an Arduino (Mega).

However, Arduino IDE fails to compile successfully, because it says it cannot find these libraries:

#include <cstddef>
#include <cstdint>
#include <cmath>
#include <array>

The <array> library seems to be a major stumbling block

Can you advise?

martinglass, 2022-02-28:
Running the sbus_example, in Arduino IDE with Arduino Mega 2560 gives this error:

_In file included from /Users/xxx/Documents/Arduino/libraries/sbus-main/examples/arduino/sbus_example/sbus_example.ino:36:0:
/Users/xxx/Documents/Arduino/libraries/sbus-main/src/sbus.h:65:8: error: 'array' in namespace 'std' does not name a template type
   std::array<int16_t, NUM_SBUS_CH_> ch_;
        ^~~~~
/Users/xxx/Documents/Arduino/libraries/sbus-main/src/sbus.h:79:15: error: 'array' in namespace 'std' does not name a template type
   inline std::array<int16_t, NUM_SBUS_CH_> ch() const {return ch_;}
               ^~~~~
/Users/xxx/Documents/Arduino/libraries/sbus-main/src/sbus.h:105:8: error: 'array' in namespace 'std' does not name a template type
   std::array<int16_t, NUM_SBUS_CH_> ch_;
        ^~~~~
/Users/xxx/Documents/Arduino/libraries/sbus-main/src/sbus.h:121:29: error: 'array' in namespace 'std' does not name a template type
   inline void ch(const std::array<int16_t, NUM_SBUS_CH_> &cmd) {ch_ = cmd;}
                             ^~~~~
/Users/xxx/Documents/Arduino/libraries/sbus-main/src/sbus.h:121:34: error: expected ',' or '...' before '<' token
   inline void ch(const std::array<int16_t, NUM_SBUS_CH_> &cmd) {ch_ = cmd;}
                                  ^
/Users/xxx/Documents/Arduino/libraries/sbus-main/src/sbus.h:122:15: error: 'array' in namespace 'std' does not name a template type
   inline std::array<int16_t, NUM_SBUS_CH_> ch() const {return ch_;}
               ^~~~~
/Users/xxx/Documents/Arduino/libraries/sbus-main/src/sbus.h: In member function 'void bfs::SbusTx::ch(int)':
/Users/xxx/Documents/Arduino/libraries/sbus-main/src/sbus.h:121:65: error: 'ch_' was not declared in this scope
   inline void ch(const std::array<int16_t, NUM_SBUS_CH_> &cmd) {ch_ = cmd;}
                                                                 ^~~
/Users/xxx/Documents/Arduino/libraries/sbus-main/src/sbus.h:121:65: note: suggested alternative: 'ch'
   inline void ch(const std::array<int16_t, NUM_SBUS_CH_> &cmd) {ch_ = cmd;}
                                                                 ^~~
                                                                 ch
/Users/xxx/Documents/Arduino/libraries/sbus-main/src/sbus.h:121:71: error: 'cmd' was not declared in this scope
   inline void ch(const std::array<int16_t, NUM_SBUS_CH_> &cmd) {ch_ = cmd;}
                                                                       ^~~
/Users/xxx/Documents/Arduino/libraries/sbus-main/src/sbus.h:121:71: note: suggested alternative: 'fmod'
   inline void ch(const std::array<int16_t, NUM_SBUS_CH_> &cmd) {ch_ = cmd;}
                                                                       ^~~
                                                                       fmod
/Users/xxx/Documents/Arduino/libraries/sbus-main/examples/arduino/sbus_example/sbus_example.ino: At global scope:
sbus_example:43:6: error: 'array' in namespace 'std' does not name a template type
 std::array<int16_t, bfs::SbusRx::NUM_CH()> sbus_data;
      ^~~~~
/Users/xxx/Documents/Arduino/libraries/sbus-main/examples/arduino/sbus_example/sbus_example.ino: In function 'void loop()':
sbus_example:57:5: error: 'sbus_data' was not declared in this scope
     sbus_data = sbus_rx.ch();
     ^~~~~~~~~
/Users/xxx/Documents/Arduino/libraries/sbus-main/examples/arduino/sbus_example/sbus_example.ino:57:5: note: suggested alternative: 'sbus_tx'
     sbus_data = sbus_rx.ch();
     ^~~~~~~~~
     sbus_tx
sbus_example:57:25: error: 'class bfs::SbusRx' has no member named 'ch'; did you mean 'ch17'?
     sbus_data = sbus_rx.ch();
                         ^~
                         ch17
exit status 1
'array' in namespace 'std' does not name a template type_

flybrianfly, 2022-02-28:
Please note from the [installation instructions](https://github.com/bolderflight/sbus#arduino) that AVR microcontrollers are not supported, mostly due to the use of std::array.

closed 2022-02-28 by flybrianfly

martinglass, 2022-02-28:
Doh, Thanks. I see that now.

Would it be possible to use arrays in a way that works for AVR devices as well as ARM?

It would be useful to AVR users, if this code could run on such devices.

flybrianfly, 2022-02-28:
There's some older versions older versions of this library that support AVR; check the checklog. Also, the [Embedded Template Library](https://github.com/ETLCPP/etl) can provide std::array support for AVR.