bolderflight/sbus #58: Support NRF52 / Arduino Nano 33 BLE

kallaspriit, 2022-07-09:
Found a way to get the library to work on the Arduino Nano 33 BLE.

By default the UART on the NRF52 does not support custom baud rates or UART config other than the default but these can be configured using registers.

Basically I did something like this:

At the start of `sbus.cpp` after the includes:
```c++
#if defined(ARDUINO_ARCH_NRF52840) // Arduino Nano 33 BLE

#define UARTE0_BASE_ADDR 0x40002000
#define UARTE1_BASE_ADDR 0x40028000
#define UART_BAUDRATE_REG_OFFSET 0x524
#define UART_CONFIG_REG_OFFSET 0x56C
#define UART_CONFIG_8E2 0x000000F
#define UART_BAUD_100000 0x19114A7
#define UART0_BAUDRATE_REGISTER (((unsigned int *)(UARTE0_BASE_ADDR + UART_BAUDRATE_REG_OFFSET)))
#define UART1_BAUDRATE_REGISTER (((unsigned int *)(UARTE1_BASE_ADDR + UART_BAUDRATE_REG_OFFSET)))
#define UART0_CONFIG_REGISTER (((unsigned int *)(UARTE0_BASE_ADDR + UART_CONFIG_REG_OFFSET)))
#define UART1_CONFIG_REGISTER (((unsigned int *)(UARTE1_BASE_ADDR + UART_CONFIG_REG_OFFSET)))

#endif
```

Added custom `Begin` signature for the NRF52840 to support both UART0 and UART1:

```c++
#elif defined(ARDUINO_ARCH_NRF52840)
    void Begin(int uart = 0);
#else
```

And then added custom initialization in `Begin`:

```c++
#elif defined(ARDUINO_ARCH_NRF52840) // Arduino Nano 33 BLE
  uart_->begin(9600);

  if (uart == 0)
  {
    *UART0_BAUDRATE_REGISTER = UART_BAUD_100000;
    *UART0_CONFIG_REGISTER = UART_CONFIG_8E2;
  }
  else if (uart == 1)
  {
    *UART1_BAUDRATE_REGISTER = UART_BAUD_100000;
    *UART1_CONFIG_REGISTER = UART_CONFIG_8E2;
  }
/* Everything else, with a hardware inverter */
#else
```

The same could be done for `SbusTx`.

Tested with Frsky R-XSR using the uninverted sbus signal from the B-pad and it works great.

[image]

Would be cool if this support was included in the library and if not then maybe it helps someone who finds this issue :)

flybrianfly, 2022-09-29:
I'm sorry, I was on travel when you sent this and it must have slipped past my radar. I'd like to implement this - is it possible to implement an inverted serial on the Nano? Does it only have two Serial ports or are there more to support? Would you be willing to test versions of the library that add support? I'm not very familiar with that platform.

kallaspriit, 2022-09-29:
Not sure about inverted but I could test the uninverted implementation. You could also pick up the board for testing, they're not expensive.

Scoobie2021, 2024-08-10:
Hi Kallaspriit, thank you very much. I am going to try your solution. I am working with a Nano 33 BLE just now.