bolderflight/sbus #59: Problems with Teensy 4.1 but not with Teensy 3.2

Embring, 2022-09-13:
Hi! Thanks for this great lib! I have noticed that **sbus_rx.Read()** lags a lot in Teensy 4.1 but works great with Teensy 3.2. I am currently using a receiver with already inverted signal Oversky XR602T-A2 running at 3.3V. No problems at all with Teensy 3.2 as said. I'm wondering if the Teensy 4.1 expects an non-already reversed input? Can I define this Option for Teensy 4.1?

[image]

Embring, 2022-09-13:
I guess I found a work around or error in the sbus.cpp file:

  /* Start the bus */
  /* Teensy 3.0 || Teensy 3.1/3.2 */
  #if defined(__MK20DX128__) || defined(__MK20DX256__)
    uart_->begin(BAUD_, SERIAL_8E1_RXINV_TXINV);
  /*
  * Teensy 3.5 || Teensy 3.6 ||
  * Teensy LC  || Teensy 4.0/4.1 ||
  * Teensy 4.0 Beta
  */
  #elif defined(__MK64FX512__) || defined(__MK66FX1M0__) || \
        defined(__MKL26Z64__)  || defined(__IMXRT1062__) || \
        defined(__IMXRT1052__)
    uart_->begin(BAUD_, SERIAL_8E2_RXINV_TXINV);  

I changed the stop bit to 1 as in the /* Teensy 3.0 || Teensy 3.1/3.2 */
   uart_->begin(BAUD_, SERIAL_8E1_RXINV_TXINV);  

And now its working!

flybrianfly, 2022-09-13:
Also, please look at the newest version on GitHub, which supports Teensy 4.1. I'm not sure what's up with the Arduino Library manager, but it doesn't always grab the newest versions.

flybrianfly, 2022-09-13:
EDIT: Nevermind, I see you changed the stop bit. Still early for me and I need coffee apparently. I'll look into it further, but haven't had issues with the Teensy 4.1 and SBUS and I fly that often.

Embring, 2022-09-13:
Great! Yes, it works very good now, when the stop bit is set to 1 like in Teensy 3.2 etc. I don't understand why there can be difference with the number of stop bits. But I notice the difference in the code. The Oversky XR602T-A2 only have one stop bit in a Oscilloscope. Attached the data for this receiver. 
[XR600T series RX manual.pdf](https://github.com/bolderflight/sbus/files/9557838/XR600T.series.RX.manual.pdf)

flybrianfly, 2022-09-13:
It's interesting, in the manual, the figures are pretty fuzzy, but it looks like the non-inverted SBUS has two stop bits whereas the inverted SBUS only has one. SBUS is supposed to be inverted with two stop bits. Teensy 3.0, 3.1, and 3.2 only support a single stop bit, which is fine for receiving and we use a workaround to use two stop bits when transmitting. I don't really want to switch to only using one stop bit, since who knows how that would affect servo compatibility. I'm chalking this up to an issue with the receiver firmware for now. I think a potential improvement for this library would be to make it easier to use non-inverted signals, either for microcontrollers that don't support inverted serial or to bypass some issues like this.

closed 2022-09-13 by flybrianfly