bolderflight/sbus #61: Does not work

timsenecal, 2022-10-28:
I have tried to use this on a Teensy LC, Teensy 3.1 and Teensy 3.2.

using your example arduino ino sketch,

sbus_rx.read() never returns true, regardless which serial hardware identifier i use, regardless of inv parameter of true or false.

I have tried this with the following sbus receivers:

Frsky x4r-sb
Frsky x8r
Frsky x6r
Jumper r8
Radiomaster r168-d16

flybrianfly, 2022-10-30:
@timsenecal, thanks, I found a bug that was introduced in our last major revision which could hang a Teensy 3.1/3.2 if the *Write* method was called. This is fixed in v8.1.4. I tested against a Teensy 3.0, Teensy 3.2 and Teensy LC and all work.

If it's still not working for you, I would check that the receiver is bound to the transmitter and the transmitter is turned on. I've noticed that the SBUS receivers don't send any packets until the transmitter has been turned on and communicating with the receiver.

tursilion, 2022-10-31:
On the Uno, found I needed to enable the input pullup to get reliable data from a X8R receiver. Perhaps similar?

timsenecal, 2022-11-01:
Just for clarification, I am using the receivers with an active bound transmitter, tested using three different models: a home grown model using a 4-in-1 transmitter module (clone d-16 mode), an FrSky Taranis x9d (internal OG accst d-16 module), and a FrSky QX7 (OG accst d-16 module).

flybrianfly, 2022-11-01:
Should be working. I would try v8.1.4 of the library. If it still isn't working, let's post some photos of our setups and maybe something will stand out.

tpsenecal, 2022-11-04:
I resolved this issue by completely deleting Arduino IDE and all of its directories from my Ubuntu machine, and re-installing everything.  am now using Arduino 1.8.19, teensyduino and sbus v8.1.4   seems to be working now.  not sure if i should try using the new Arduino 2.0.1 IDE or not.

flybrianfly, 2022-11-06:
Glad to hear it’s working. 

I would recommend holding off on Arduino 2.x for now. Teensyduino support is currently in Beta and I haven’t done any testing of BFS libraries against Arduino 2.x yet. I’m waiting for a stable Teensyduino release before I’ll work with Arduino 2.x on my end.

closed 2022-11-06 by flybrianfly