bolderflight/sbus #63: FrSky 24 Channels

rsaleev, 2022-11-16:
I have Tandem X18 and it's possible to use 24 channels instead of 16/17/18. Do anyone have an idea how to get values from 24th  byte?

flybrianfly, 2022-11-16:
Unfortunately, I don't see much documentation yet on how the 24 channel data works. Might be a bit to get this updated until more people get their hands on them and reverse engineer the protocol changes. Do you know if channels 16 - 24 are simply on / off commands or are they full-range like sticks and sliders?

rsaleev, 2022-11-18:
Well, I am.a newbie in C and microcontrollers. 
Tandem X18 is a guy customized device. Channel 16-24 can be a button, a switch or even stick.
Does reverse engineering means that I have to test one by one each parameter and compare it to bit fields values? If so, I know the right path to achieve the goal.

flybrianfly, 2022-11-18:
Given that those channels are more than just bits, they're either compressing the other channels or using more than the standard 25 bytes for a packet. It would be very difficult to reverse engineer as a newbie.

flybrianfly, 2025-04-09:
> any progress on this?

@dodds-steven, no, I haven't had a chance to look into it yet. There's a link chain in Issue #73 that will take you to some code that looks like it would be a good starting point.