bolderflight/sbus #66: ESP32 Compilation Error

Dr-Drone, 2023-03-06:
Hi, I'm trying to compile for ESP32 s2 and use pin 14 as SBUS input to display sbus output. I'm using the sbus_example script as is and it isn't working. Any help would be greatly appreciated. Thanks!

Error: sbus_example.ino:30:22: error: 'Serial2' was not declared in this scope
 bfs::SbusRx sbus_rx(&Serial2);
                      ^~~~~~~

flybrianfly, 2023-03-06:
Looks like ESP32 doesn't have Serial2. You need to pass the correct Serial bus to the SBUS constructor. Also, please review the [library documentation for ESP32](https://github.com/bolderflight/sbus#sbusrx). You'll see that the constructor for ESP32 requires the pointer to the serial bus object along with the pin numbers and a boolean specifying whether to use an inverted SBUS signal.

closed 2023-03-06 by flybrianfly

Dr-Drone, 2023-03-06:
Thanks a lot! Got it to work. Had to use 3.3v logic and updated the following for the esp32 s2 mini 

#include "sbus.h"
#include <HardwareSerial.h>

HardwareSerial SBUSS(1);

/* SBUS object, reading SBUS */
bfs::SbusRx sbus_rx(&SBUSS, 14, 15, true);
/* SBUS object, writing SBUS */
bfs::SbusTx sbus_tx(&SBUSS, 14, 15, true);
/* SBUS data */
bfs::SbusData data;