bolderflight/sbus #67: Teensy 3.2 Compilation error

Duese123, 2023-04-16:
Hello wanted to update my project my lib for sbus was on 2.x.x 
i now get on Arduiono 1.8.5 and as well >2.x the following error 
( i installed teensy via Arduino IDE 2 with adding the link https://www.pjrc.com/teensy/package_teensy_index.json)

c:/users/xx/appdata/local/arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\xx\AppData\Local\Temp\arduino_modified_sketch_862059/sketch_apr16a.ino:44: undefined reference to `bfs::SbusTx::Begin()'
c:/users/xx/appdata/local/arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\xx\AppData\Local\Temp\arduino_build_556759\sketch\sketch_apr16a.ino.cpp.o: in function `loop':
C:\Users\xx\AppData\Local\Temp\arduino_modified_sketch_862059/sketch_apr16a.ino:48: undefined reference to `bfs::SbusRx::Read()'
c:/users/xx/appdata/local/arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\xx\AppData\Local\Temp\arduino_modified_sketch_862059/sketch_apr16a.ino:63: undefined reference to `bfs::SbusTx::Write()'
collect2.exe: error: ld returned 1 exit status

not sure where the issue is

flybrianfly, 2023-04-16:
Are you on the latest version of this library downloaded from GitHub?

Duese123, 2023-04-16:
yes and calling specificly the version with 
#include <C:\Users\xx\Documents\Arduino\libraries\sbus-8.1.4\src\sbus.h

flybrianfly, 2023-04-17:
Those warnings aren't really bad, but you definitely are not using the most current version of the library. The most current version includes fixes to eliminate the warnings.

Duese123, 2023-04-17:
yep you were right. Seems that Arduino was not using the correct one. it showed:
[image]

i downloaded again the zip and overwrote all data inside the lib now all errors are gone.

Strange that it was not working out of the box with downloading and inserting in libraries folder. i had to go the way through Arduino IDE. I have to say i have installed 1.8.15 and newly 2.x in parallel.

Thank you by the way for your fast reply

closed 2023-04-17 by Duese123