bolderflight/sbus #69: Frsky D8 SoftwareSerial

giubu, 2023-05-07:
Hello, thank you for your hard work.
I am trying to set up a D8 protocol receiver with a NANO. 
I would like to use SoftwareSerial, so as to debug via USB.
is this possible?
thank you

flybrianfly, 2023-05-07:
It would definitely take some modifications since the library is built around the HardwareSerial interface. SBUS is at a baud rate of 100000, so also not sure if it's feasible to use software serial from a computing standpoint. Maybe it would be better to use the HardwareSerial for SBUS and the software serial to output debug info?

giubu, 2023-05-08:
Thank you for the reply. That is a very good suggestion, and I have also seen it in another SBUS library. But I have also seen SoftwareSerial used for SBUS. If it can be done, it would be great for developing extra functionalities on the bench.
By the way, everything else is working fine on an Arduino NANO. Thank you

giubu, 2023-05-08:
I am exploring possibilities of using AltSoftSerial library, which apparently can do 100000 baud rate. Could you give me an indication on how to modify the code? Do I have to implement it in the sbus.h file in /src ?
thank you