bolderflight/sbus #70: Using this library, there was an error running on esp32 arduino

changlin1990, 2023-06-12:
When running esp32 on the arduino ide and using this library, an error occurred. Below is a screenshot of the error prompt. Please help identify and fix the problem.
[image]
[image]
[image]

flybrianfly, 2023-06-12:
Compiles fine for me using Arduino 1.8.19, the latest version of the library downloaded directly from GitHub, and compiling for ESP32 Dev. Please confirm that you are using the latest version of the library downloaded directly from GitHub (Arduino's library manager sometimes doesn't catch version updates). If you're still encountering issues, please provide the Arduino version number and the exact board that you are compiling for as well as the version of the ESP32 board software.

changlin1990, 2023-06-12:
> Compiles fine for me using Arduino 1.8.19, the latest version of the library downloaded directly from GitHub, and compiling for ESP32 Dev. Please confirm that you are using the latest version of the [...]

> 使用 Arduino 1.8.19 编译得很好，Arduino 32.32.<> 是直接从 GitHub 下载的最新版本的库，并为 ESP<> Dev 编译。请确认您使用的是直接从 GitHub 下载的最新版本的库（Arduino 的库管理器有时无法捕获版本更新）。如果您仍然遇到问题，请提供 Arduino 版本号和您正在编译的确切开发板以及 ESP<> 开发板软件的版本。

changlin1990, 2023-06-12:
I am using the latest version, please take a screenshot, the IDE is 2.1
[image]

changlin1990, 2023-06-12:
Do you have these error prompts when trying out version 2.0 of the Arduino IDE?

flybrianfly, 2023-06-13:
No, still compiles without errors or warnings. Again, recommend downloading and installing directly from GitHub rather than the builtin Arduino library manager.

[image]

changlin1990, 2023-06-13:
好的，谢谢，我再试一试，

OK, thank you. I'll try again.

changlin1990, 2023-06-13:
I manually installed the latest library, but there was still an error. The error message is as follows. What are the possible reasons?
[image]

Rytiggy, 2024-01-04:
I was able to get it to compile by using the ESP32 dev module for the board.
[image]

Rytiggy, 2024-01-04:
I was able to get this working, I only needed to use the RX input so the code chages to:

> Note that I am using the RX 2 pin on the ESP 32. which is GPIO 16.

```C
#include "sbus.h"

/* SBUS object, reading SBUS */
// bfs::SbusRx sbus_rx(&Serial2, 26, 27, 1, 1);
bfs::SbusRx sbus_rx(&Serial2, 16, 17, true);
/* SBUS object, writing SBUS */
// bfs::SbusTx sbus_tx(&Serial2, 26, 27, 1, 1);
/* SBUS data */
bfs::SbusData data;

void setup() {
  /* Serial to display data */
  Serial.begin(115200);
  while (!Serial) {}

  delay(2000);

  /* Begin the SBUS communication */
  Serial.println("Begin SBUS...");
  sbus_rx.Begin();
  // sbus_tx.Begin();
}

void loop() {

  if (sbus_rx.Read()) {
    /* Grab the received data */
    data = sbus_rx.data();
    /* Display the received data */
    for (int8_t i = 0; i < data.NUM_CH; i++) {
      Serial.print(data.ch[i]);
      Serial.print("\t");
    }
    
    /* Display lost frames and failsafe data */
    Serial.print(data.lost_frame);
    Serial.print("\t");
    Serial.println(data.failsafe);
    /* Set the SBUS TX data to the received data */
    // sbus_tx.data(data);
    /* Write the data to the servos */
    // sbus_tx.Write();
  }
}

```

 ### Example
[image]

Hope this helps

blzhu, 2024-01-21:
I also can work on esp32 board with Arduino IDE 2.0，but can not work for esp8266

iurycarlos, 2024-05-09:
are you using a 5 to 3.3v converter?

Witty-Wizard, 2024-07-02:
https://github.com/Witty-Wizard/SerialIO

Try using this library