bolderflight/sbus #71: Issues during compiling for ESP32 in Arduino IDE 1.8.19

themanfrommoon, 2023-07-21:
Hello,

I have the Arduino IDE 1.8.19 and downloaded the latest .zip library sbus-main.zip
I unzipped it to C:\Users\Chris\Documents\Arduino\libraries
I started the IDE, choosed ESP32 Dev Module and tried to compile.
But it shows following error messages.

Any idea what is wrong here?
How to solve the problem?

`sbus_example:29:29: error: no matching function for call to 'bfs::SbusRx::SbusRx(HardwareSerial*)'
 bfs::SbusRx sbus_rx(&Serial2);
                             ^
In file included from C:\Users\Chris\Documents\Arduino\libraries\sbus-main\examples\arduino\sbus_example\sbus_example.ino:26:
C:\Users\Chris\Documents\Arduino\libraries\sbus-main\src/sbus.h:53:3: note: candidate: 'bfs::SbusRx::SbusRx(HardwareSerial*, int8_t, int8_t, bool, bool)'
   SbusRx(HardwareSerial *bus, const int8_t rxpin, const int8_t txpin,
   ^~~~~~
C:\Users\Chris\Documents\Arduino\libraries\sbus-main\src/sbus.h:53:3: note:   candidate expects 5 arguments, 1 provided
C:\Users\Chris\Documents\Arduino\libraries\sbus-main\src/sbus.h:50:3: note: candidate: 'bfs::SbusRx::SbusRx(HardwareSerial*, int8_t, int8_t, bool)'
   SbusRx(HardwareSerial *bus, const int8_t rxpin, const int8_t txpin,
   ^~~~~~
C:\Users\Chris\Documents\Arduino\libraries\sbus-main\src/sbus.h:50:3: note:   candidate expects 4 arguments, 1 provided
C:\Users\Chris\Documents\Arduino\libraries\sbus-main\src/sbus.h:47:7: note: candidate: 'constexpr bfs::SbusRx::SbusRx(const bfs::SbusRx&)'
 class SbusRx {
       ^~~~~~
C:\Users\Chris\Documents\Arduino\libraries\sbus-main\src/sbus.h:47:7: note:   no known conversion for argument 1 from 'HardwareSerial*' to 'const bfs::SbusRx&'
C:\Users\Chris\Documents\Arduino\libraries\sbus-main\src/sbus.h:47:7: note: candidate: 'constexpr bfs::SbusRx::SbusRx(bfs::SbusRx&&)'
C:\Users\Chris\Documents\Arduino\libraries\sbus-main\src/sbus.h:47:7: note:   no known conversion for argument 1 from 'HardwareSerial*' to 'bfs::SbusRx&&'
sbus_example:31:29: error: no matching function for call to 'bfs::SbusTx::SbusTx(HardwareSerial*)'
 bfs::SbusTx sbus_tx(&Serial2);
                             ^
In file included from C:\Users\Chris\Documents\Arduino\libraries\sbus-main\examples\arduino\sbus_example\sbus_example.ino:26:
C:\Users\Chris\Documents\Arduino\libraries\sbus-main\src/sbus.h:107:3: note: candidate: 'bfs::SbusTx::SbusTx(HardwareSerial*, int8_t, int8_t, bool, bool)'
   SbusTx(HardwareSerial *bus, const int8_t rxpin, const int8_t txpin,
   ^~~~~~
C:\Users\Chris\Documents\Arduino\libraries\sbus-main\src/sbus.h:107:3: note:   candidate expects 5 arguments, 1 provided
C:\Users\Chris\Documents\Arduino\libraries\sbus-main\src/sbus.h:104:3: note: candidate: 'bfs::SbusTx::SbusTx(HardwareSerial*, int8_t, int8_t, bool)'
   SbusTx(HardwareSerial *bus, const int8_t rxpin, const int8_t txpin,
   ^~~~~~
C:\Users\Chris\Documents\Arduino\libraries\sbus-main\src/sbus.h:104:3: note:   candidate expects 4 arguments, 1 provided
C:\Users\Chris\Documents\Arduino\libraries\sbus-main\src/sbus.h:101:7: note: candidate: 'constexpr bfs::SbusTx::SbusTx(const bfs::SbusTx&)'
 class SbusTx {
       ^~~~~~
C:\Users\Chris\Documents\Arduino\libraries\sbus-main\src/sbus.h:101:7: note:   no known conversion for argument 1 from 'HardwareSerial*' to 'const bfs::SbusTx&'
C:\Users\Chris\Documents\Arduino\libraries\sbus-main\src/sbus.h:101:7: note: candidate: 'constexpr bfs::SbusTx::SbusTx(bfs::SbusTx&&)'
C:\Users\Chris\Documents\Arduino\libraries\sbus-main\src/sbus.h:101:7: note:   no known conversion for argument 1 from 'HardwareSerial*' to 'bfs::SbusTx&&'
Bibliothek sbus-main in Version 8.1.4 im Ordner: C:\Users\Chris\Documents\Arduino\libraries\sbus-main  wird verwendet
exit status 1
no matching function for call to 'bfs::SbusRx::SbusRx(HardwareSerial*)'
`

Thanks a lot and best regards,
Chris

flybrianfly, 2023-07-22:
Hi Chris,

Please note that due to the *unique* way that ESP32 handles serial ports, the object declaration is different, so the examples won't work as is. Documentation is here:
https://github.com/bolderflight/sbus#sbusrx

An example would be:

```C++
/* SBUS object, reading SBUS */
bfs::SbusRx sbus_rx(&Serial1, 27, 26, true);
/* SBUS object, writing SBUS */
bfs::SbusTx sbus_tx(&Serial1, 27, 26, true);
```

Where *Serial1* is a reference to the Serial1 object, 27 is the RX pin, 26 is the TX pin, and we're using inverted logic.

themanfrommoon, 2023-07-23:
Hi Brian,

thanks a lot, this solved the issue.
I was refering to a video I found: https://www.youtube.com/watch?v=IhlXGybLxcc&t=593s
But it seems that there is a difference in the definition.

However, thanks a lot! It works fine now!

Best regards,
Chris

closed 2023-07-23 by themanfrommoon