bolderflight/sbus #74: Fix warnings from compiler to do with brackets please

MoveElectricMobility, 2024-01-15:
Here are the following warning to fix please:

``c:\Users\chris\OneDrive\Documents\Arduino\libraries\Bolder_Flight_Systems_SBUS\src\sbus.cpp: In member function 'bool bfs::SbusRx::Parse()':
c:\Users\chris\OneDrive\Documents\Arduino\libraries\Bolder_Flight_Systems_SBUS\src\sbus.cpp:112:68: warning: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
  112 |         data_.ch[0]  = static_cast<int16_t>(buf_[1] | buf_[2] << 8 & 0x07FF);
      |                                                       ~~~~~~~~~~~~~^~~~~~~~
c:\Users\chris\OneDrive\Documents\Arduino\libraries\Bolder_Flight_Systems_SBUS\src\sbus.cpp:113:73: warning: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
  113 |         data_.ch[1]  = static_cast<int16_t>(buf_[2] >> 3 | buf_[3] << 5 &
      |                                                            ~~~~~~~~~~~~~^
  114 |                                             0x07FF);
      |                                             ~~~~~~                       
c:\Users\chris\OneDrive\Documents\Arduino\libraries\Bolder_Flight_Systems_SBUS\src\sbus.cpp:116:59: warning: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
  116 |                                             buf_[5] << 10 & 0x07FF);
      |                                             ~~~~~~~~~~~~~~^~~~~~~~
c:\Users\chris\OneDrive\Documents\Arduino\libraries\Bolder_Flight_Systems_SBUS\src\sbus.cpp:117:73: warning: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
  117 |         data_.ch[3]  = static_cast<int16_t>(buf_[5] >> 1 | buf_[6] << 7 &
      |                                                            ~~~~~~~~~~~~~^
  118 |                                             0x07FF);
      |                                             ~~~~~~                       
c:\Users\chris\OneDrive\Documents\Arduino\libraries\Bolder_Flight_Systems_SBUS\src\sbus.cpp:119:73: warning: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
  119 |         data_.ch[4]  = static_cast<int16_t>(buf_[6] >> 4 | buf_[7] << 4 &
      |                                                            ~~~~~~~~~~~~~^
  120 |                                             0x07FF);
      |                                             ~~~~~~                       
c:\Users\chris\OneDrive\Documents\Arduino\libraries\Bolder_Flight_Systems_SBUS\src\sbus.cpp:122:58: warning: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
  122 |                                             buf_[9] << 9 & 0x07FF);
      |                                             ~~~~~~~~~~~~~^~~~~~~~
c:\Users\chris\OneDrive\Documents\Arduino\libraries\Bolder_Flight_Systems_SBUS\src\sbus.cpp:123:74: warning: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
  123 |         data_.ch[6]  = static_cast<int16_t>(buf_[9] >> 2 | buf_[10] << 6 &
      |                                                            ~~~~~~~~~~~~~~^
  124 |                                             0x07FF);
      |                                             ~~~~~~                        
c:\Users\chris\OneDrive\Documents\Arduino\libraries\Bolder_Flight_Systems_SBUS\src\sbus.cpp:125:75: warning: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
  125 |         data_.ch[7]  = static_cast<int16_t>(buf_[10] >> 5 | buf_[11] << 3 &
      |                                                             ~~~~~~~~~~~~~~^
  126 |                                             0x07FF);
      |                                             ~~~~~~                         
c:\Users\chris\OneDrive\Documents\Arduino\libraries\Bolder_Flight_Systems_SBUS\src\sbus.cpp:127:70: warning: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
  127 |         data_.ch[8]  = static_cast<int16_t>(buf_[12] | buf_[13] << 8 & 0x07FF);
      |                                                        ~~~~~~~~~~~~~~^~~~~~~~
c:\Users\chris\OneDrive\Documents\Arduino\libraries\Bolder_Flight_Systems_SBUS\src\sbus.cpp:128:75: warning: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
  128 |         data_.ch[9]  = static_cast<int16_t>(buf_[13] >> 3 | buf_[14] << 5 &
      |                                                             ~~~~~~~~~~~~~~^
  129 |                                             0x07FF);
      |                                             ~~~~~~                         
c:\Users\chris\OneDrive\Documents\Arduino\libraries\Bolder_Flight_Systems_SBUS\src\sbus.cpp:131:60: warning: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
  131 |                                             buf_[16] << 10 & 0x07FF);
      |                                             ~~~~~~~~~~~~~~~^~~~~~~~
c:\Users\chris\OneDrive\Documents\Arduino\libraries\Bolder_Flight_Systems_SBUS\src\sbus.cpp:132:75: warning: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
  132 |         data_.ch[11] = static_cast<int16_t>(buf_[16] >> 1 | buf_[17] << 7 &
      |                                                             ~~~~~~~~~~~~~~^
  133 |                                             0x07FF);
      |                                             ~~~~~~                         
c:\Users\chris\OneDrive\Documents\Arduino\libraries\Bolder_Flight_Systems_SBUS\src\sbus.cpp:134:75: warning: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
  134 |         data_.ch[12] = static_cast<int16_t>(buf_[17] >> 4 | buf_[18] << 4 &
      |                                                             ~~~~~~~~~~~~~~^
  135 |                                             0x07FF);
      |                                             ~~~~~~                         
c:\Users\chris\OneDrive\Documents\Arduino\libraries\Bolder_Flight_Systems_SBUS\src\sbus.cpp:137:59: warning: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
  137 |                                             buf_[20] << 9 & 0x07FF);
      |                                             ~~~~~~~~~~~~~~^~~~~~~~
c:\Users\chris\OneDrive\Documents\Arduino\libraries\Bolder_Flight_Systems_SBUS\src\sbus.cpp:138:75: warning: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
  138 |         data_.ch[14] = static_cast<int16_t>(buf_[20] >> 2 | buf_[21] << 6 &
      |                                                             ~~~~~~~~~~~~~~^
  139 |                                             0x07FF);
      |                                             ~~~~~~                         
c:\Users\chris\OneDrive\Documents\Arduino\libraries\Bolder_Flight_Systems_SBUS\src\sbus.cpp:140:75: warning: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
  140 |         data_.ch[15] = static_cast<int16_t>(buf_[21] >> 5 | buf_[22] << 3 &
      |                                                             ~~~~~~~~~~~~~~^
  141 |                                             0x07FF);
      |                                             ~~~~~~  ``

flybrianfly, 2024-01-15:
Please use the newest version of this library, downloaded directly from GitHub - those errors have been addressed.

closed 2024-01-15 by flybrianfly

MoveElectricMobility, 2024-01-16:
I am using 8.1.4 from the arduino library manager which is the newest version, if things have changed since then maybe push a 8.1.5 version?