bolderflight/sbus #77: Can Recv Sbus data with ESP8266

blzhu, 2024-01-21:
I want to use ESP8266 to handle sbus data, UART0 for code download and SBus RX, and Debug info output in a softwareSerial,  build ok, but sbus_rx.read always false,  it is ok for ESP32 use two hardware Serial

bfs::SbusRx sbus_rx(&Serial)
sbus_rx.read == false 

board: WeMos D1 R2 mini esp8266
Arduino IDE 2.2.1

blzhu, 2024-01-22:
I change source code , use [espsoftwareSerial] as SBus RX, and UART0 as debug output, it works , and need change CPU freq at 160Mhz, use default setting 80Mhz will have error data recv,  this way will downgrade CPU performence, not a good solution, it better find root issue  ,let UART0 as sbus Rx

Eisbaeeer, 2024-06-21:
> I change source code , use [espsoftwareSerial] as SBus RX, and UART0 as debug output, it works , and need change CPU freq at 160Mhz, use default setting 80Mhz will have error data recv, this way will downgrade CPU performence, not a good solution, it better find root issue ,let UART0 as sbus Rx

Can you provide your changes reagarding softwareserial?
Changing CPU freq. is already done.
Thanks.