bolderflight/sbus #79: Unable to receive from Futaba R7003SB

ongbh, 2024-10-28:
Hi Hi,
I tested the library and code with a few devices, including 2 ELRS RX (Happymodel EPW6, Matek R24-P6), flysky FS-iA10B, and a ToolkitRC MC8, however, I just can't get any reading with a Futaba R7003SB.

Any advice on what's best to troubleshooting? I am using a Seed Studio Xiao ESP32.

flybrianfly, 2024-10-28:
It's probably [this line](https://github.com/bolderflight/sbus/blob/main/src/sbus.cpp#L99) causing an issue. The code is checking that the current byte is the header and the previous byte is the footer. We do that since SBUS only uses a 1 byte header, which could also be found in the payload, so this is a way to give a little more assurance that we're really looking at the header byte.

I thought I had the correct footer for SBUS2 as an option, but I'm not seeing it in your data capture.

ongbh, 2024-10-29:
I going to close this case first. Because when I switch it to FASTest-14CH on my T14SG, the R7003SB can decode okie.

closed 2024-10-29 by ongbh